gbroques/ose-3d-printer-workbench #1: Add Angle Frame Connector to Frame Model

gbroques, 2020-02-15:
The [Angle Frame Connector](https://wiki.opensourceecology.org/wiki/Angle_Frame_Connector) is the new standard for building D3D Pro printer frames as it eliminates the need for welding or gluing frame pieces together with epoxy.

The top corners have other corner pieces to which the axes are attached to (see red and orange images below or on [wiki](https://wiki.opensourceecology.org/wiki/Angle_Frame_Connector)).

Currently our frame model appears as if it were welded.

A basic model for the angle frame connector has been created inside [angle_frame_connector.py](https://github.com/gbroques/ose-3d-printer-workbench/blob/master/ose3dprinter/core/model/frame/angle_frame_connector.py), but is not yet incorporated in the [frame model](https://github.com/gbroques/ose-3d-printer-workbench/blob/master/ose3dprinter/core/model/frame/frame_model.py).

### Images
[image]

[image]

[image]

marcin-ose, 2020-02-29:
The angle frame connectors that hang the Y axis have holes that stretch outside of the volume of the frame. The Y axes mount to these holes. This means that the Y axis is slightly longer than the frame size, by the amount that the holes extend beyond the edge of the frame.

marcin-ose, 2020-02-29:
Observe the heated bed rod mounts - https://wiki.opensourceecology.org/wiki/File:Carriage_idler_merge.fcstd - which hold the rods that hold the bed up. The exact length of the rods that hold the bed is such that when the rods lie in these rod holders, the rods do not interfere with the Z axis rods. That is, the bed holder rods are exactly the inner length of the frame.

marcin-ose, 2020-02-29:
The X axis rests on https://wiki.opensourceecology.org/wiki/File:Carriage_motor_merge.fcstd on the motor side, and on https://wiki.opensourceecology.org/wiki/File:Carriage_halfcarriage_merge.fcstd on the idler side. The latter is a mechanism for auto paralleling of y, such that the X rods go into it and can slide in and out if the Y axes are not parallel. The rods go all the way into the https://wiki.opensourceecology.org/wiki/File:Carriage_motor_merge.fcstd and https://wiki.opensourceecology.org/wiki/File:Carriage_halfcarriage_merge.fcstd , which can be seen in the CAD of the gantry - https://wiki.opensourceecology.org/wiki/File:D3d1911.fcstd from Feb 28, 2020.

gbroques, 2020-03-21:
> Observe the heated bed rod mounts - https://wiki.opensourceecology.org/wiki/File:Carriage_idler_merge.fcstd - which hold the rods that hold the bed up. The exact length of the rods that hold the bed [...]

This comment was broken out into issue #14.

gbroques, 2020-03-21:
> The X axis rests on https://wiki.opensourceecology.org/wiki/File:Carriage_motor_merge.fcstd on the motor side, and on https://wiki.opensourceecology.org/wiki/File:Carriage_halfcarriage_merge.fcstd [...]

This comment was broken out into issue #13.

commit e31cbbfe (2020-03-21): Add angle frame connector to frame model (Closes #1)
commit b21735f7 (2020-03-21): Add angle frame connector to frame model (Closes #1)
commit c0883ded (2020-03-21): Add angle frame connector to frame model (Closes #1)
commit b5747201 (2020-03-21): Add angle frame connector to frame model (Closes #1)
closed 2020-03-21 by gbroques