gbroques/ose-3d-printer-workbench #9: Add to FreeCAD-addons repo?

luzpaz, 2020-03-02:
https://github.com/FreeCAD/FreeCAD-addons

gbroques, 2020-03-02:
Hi @luzpaz, thanks for suggesting this idea!

I'd like to add the workbench to the FreeCAD-addons repo once the workbench is further developed and polished--likely in a couple more months once summer starts.

This currently is not a high-priority item for me as FreeCAD 16 is my main targeted user-base in which the Addon Manager is unavailable.

luzpaz, 2020-03-02:
I see. Ok. good luck then.

gbroques, 2020-04-01:
We're working towards an initial `v0.1.0` release.

See [the v0.1.0 milestone](https://github.com/gbroques/ose-3d-printer-workbench/milestone/1) for progress.

I'm planning on addressing this after that milestone is achieved.

gbroques, 2020-04-23:
@luzpaz See related PRs to add this workbench to the FreeCAD-addons repo:
* https://github.com/FreeCAD/FreeCAD-addons/pull/167
* https://github.com/FreeCAD/FreeCAD/pull/3379

gbroques, 2020-04-25:
**TODO:** We should add a status badge to the README once this is complete like the following:

[image]

gbroques, 2020-06-07:
https://github.com/FreeCAD/FreeCAD-addons/pull/167

closed 2020-06-07 by gbroques