gbroques/ose-3d-printer-workbench #99: Install problems

iplayfast, 2020-10-27:
After installing using freecad, it crashes because OSECore isn't installed. 
Using conda to install OSCCore doesn't work, (some conflict in my conda with cuda).
using git clone and moving OSCCore to the 3d workbench directory in Mod allows the menu to show, but selecting the menu seems to die.

I think you need a better install system.

gbroques, 2020-11-02:
> After installing using freecad, it crashes because OSECore isn't installed.

Yes, `ose-workbench-core` is a dependency for `ose-3d-printer-workbench`.

This is stated under [Installation](https://github.com/gbroques/ose-3d-printer-workbench#installation) in the **README**.

> Follow step 2 for [ose-workbench-core](https://github.com/gbroques/ose-workbench-core) by downloading or cloning into the same directory (e.g. ~/.FreeCAD/Mod/ose-workbench-core).

---

> Using conda to install OSCCore doesn't work, (some conflict in my conda with cuda).

Please provide more detail such as an **exact** error message or stack-trace..

---

> using git clone and moving OSCCore to the 3d workbench directory in Mod allows the menu to show, but selecting the menu seems to die.

Is there any error message displayed in the terminal if you run FreeCAD from the command line, or within [FreeCAD's report view](https://wiki.freecadweb.org/Report_view)?

---

> I think you need a better install system.

Thanks. Open to PRs and suggestions. :) 

---

Additionally, what operating system are you using? I've only really tested in **Ubuntu** / Linux.

gbroques, 2020-11-02:
> **using git clone and moving OSCCore to the 3d workbench directory in Mod** ...

You need to move `ose-workbench-core` to the `Mod/` directory.

Do not place it inside the `ose-3d-printer-workbench/` directory.

iplayfast, 2020-11-03:
That was part of the issue. The other part was a bad cuda install. (thanks NVIDIA). All has been resolved so now for the fun! Thanks for the help.

gbroques, 2020-11-12:
> That was part of the issue. The other part was a bad cuda install. (thanks NVIDIA). All has been resolved so now for the fun! Thanks for the help.

Happy to help! Thanks for the interest in the project.

Closing this issue.

closed 2020-11-12 by gbroques